ts1/BLEUnlock #3: Get notified on iPhone/iWatch when mac is locked/unlocked

wenmin-wu, 2019-11-01:
Hi, this project is quite useful. However I’m still wondering is it possible to get notified on iPhone/iWatch when mac is locked/unlocked.

ts1, 2019-11-02:
Hi @wenmin-wu 
Unfortunately that feature requires a companion app on iPhone/Watch side.
And creating iPhone/Watch app is beyond this project at the moment.

wenmin-wu, 2019-11-02:
@ts1 ok, I see, thanks. By the way, is there any way to get the device name like "William's iPhone". The shown device name while scanning is hard to recognize which one is mine.

ts1, 2019-11-02:
That information is not accessible through the official API of macOS.
Maybe it is hidden for privacy.
I'd like to know if someone knows how.

closed 2020-02-16 by ts1

nightwallaby, 2021-04-18:
@wenmin-wu I use the script to send a Pushover notification to my phone which works well.