ts1/BLEUnlock #9: [question] BLE UUID broadcast if connected?

leonardpitzu, 2020-03-20:
i have a garmin fenix 5 smartwatch which i want to use also to unlock my macbook
it works great IF the watch is not connected to my phone. if it's connected to the phone BLEunlock does not "see" the watch any more thus is unable to unlock the computer.
are you aware of any other device-unique ID which is still broadcasted even if the watch is connected to the phone (or a way to see all BLE broadcasted data so maybe i can fine one on my own)?
if this would work then the garmin would be a full replacement of the apple watch.

ts1, 2020-03-21:
BLEUnlock displays all the device IDs received and made available via CoreBluetooth API.
If your watch is not displayed, there is nothing I can do.

leonardpitzu, 2020-03-23:
ok, i assumed it to be like this but i though a question worth asking as i don't know bluetooth related stuff so well.

closed 2020-03-23 by leonardpitzu