ts1/BLEUnlock #10: Option to disable Unlock

idominiki, 2020-03-22:
Would be great If I can disable the unlock option like the lock option. 

Thanks for your work!

ts1, 2020-03-23:
@idominiki I'll add this option when I have time.

ts1, 2020-04-12:
Done.

timmerk, 2020-04-12:
I saw you did this in the latest release - thank you!

idominiki, 2020-04-18:
Hey @ts1 
even If I disable the unlock, BLEunlock want to enter a password in login screen. If no password is set, it shows a message that no password is set after login.

Can look at this or help?

Thanks and Greets
[image]

ts1, 2020-04-19:
@idominiki 
I'll fix it soon.

commit c9561514 (2020-04-21): Don't ask for password when unlock is disabled

Closes #10
closed 2020-04-21 by ts1