ts1/BLEUnlock #12: how can i find which is my device

arjing, 2020-05-21:
[image]

ts1, 2020-05-21:
Raise minimum RSSI to -60 or so and keep your device close to Mac.
Most probably the one with highest RSSI is your device.

chenrylee, 2020-09-08:
@ts1 Actually not, here's a test:
I put 2 iPhones (iPhone XR and iPhone 6S Plus) near my laptop, XR is tightly close to my laptop(0mm), and 6SP is about 1 meter, but the RSSI of 6SP(-36dBm) is higher than the XR (-44dBm).

ts1, 2020-09-19:
@arjing Please try https://github.com/ts1/BLEUnlock/releases/tag/v1.9.0-beta
It should display Bluetooth device name.

ts1, 2020-10-15:
@chenrylee it may be because transmitting signal strength is different per device.
Anyways, in version 1.9.0 it displays Bluetooth name and MAC address of devices.
I believe this fixed the issue.
Feel free to reopen if you still have problems.

closed 2020-10-15 by ts1