ts1/BLEUnlock #13: Can not detect any device Intermittent.

brilliantwf, 2020-05-28:
MacOS 10.14.6
BLEUnlock 1.8.2

ts1, 2020-05-28:
Can you explain your problem in more detail?
What is the Mac model, what is the BLE device, what is the expectation and what is the reality?

brilliantwf, 2020-05-28:
Mac model:Macbook pro 2017
Expectation:Scan and detecte my Bluetooth device(iPhoneXR)
Reality:Device List is empty(I need to restart the Bluetooth service or restart the laptop before my device could appear)

ts1, 2020-05-28:
Thank you for the details.
This happens occasionally in my environment as well.
It feels like it's become more so, especially since iOS 13.

In my case, the following solves the problem:
- Turn off and turn on Bluetooth on the Mac
- Rebooting the Mac
- Restarting the iPhone

If that doesn't work, try the following:
- Switch the Passive Mode of the BLEUnlock
- Upgrading macOS to the latest

ts1, 2020-05-30:
This happened to me today, and the only solution was rebooting iPhone.

ts1, 2020-09-19:
@brilliantwf I added option to set no-signal timeout. If you set it longer, the chance to lose signal should be less.
Please try https://github.com/ts1/BLEUnlock/releases/tag/v1.9.0-beta

ts1, 2020-10-15:
I believe this issue has been fixed in version 1.9.0, which has more robust active mode.
Feel free to reopen if you still have problems.

closed 2020-10-15 by ts1