ts1/BLEUnlock #17: Display Bluetooth name instead of device name

VFiee, 2020-09-16:
My device is Iphone xs. There are multiple people in the company who use Iphone xs. In order to avoid the wrong choice, I modified my Bluetooth name. However, when I search for the device again, I cannot find my device (the modified Bluetooth name, even though 
You can check the distance through dBm, but it is usually easy to confuse), so if you modify it to display according to the Bluetooth name, it will be very helpful to people like me, thank you!

ts1, 2020-09-19:
I found a way.
This feature will be released shortly!

ts1, 2020-09-19:
@VFiee Please try https://github.com/ts1/BLEUnlock/releases/tag/v1.9.0-beta

VFiee, 2020-09-19:
> @VFiee Please try https://github.com/ts1/BLEUnlock/releases/tag/v1.9.0-beta
Okay, I'll try it later and give you feedback. thank you!

VFiee, 2020-09-19:
> > @VFiee Please try https://github.com/ts1/BLEUnlock/releases/tag/v1.9.0-beta
> > Okay, I'll try it later and give you feedback. thank you!

It seems that the expected behavior does not occur (the Bluetooth name is displayed)!
Show a few pictures,
[image]
[image]
[image]

ts1, 2020-09-19:
@VFiee Thanks for reporting!
On my Mac, it's displayed.
Maybe, if you pair your iPhone and Mac on Bluetooth Preference, it should be displayed.
I'll think about it some more.
Thanks anyway!

VFiee, 2020-09-19:
> @VFiee Thanks for reporting!
> On my Mac, it's displayed.
> Maybe, if you pair your iPhone and Mac on Bluetooth Preference, it should be displayed.
> I'll think about it some more.
> Thanks anyway!

I just tried it again, first quit BleUnlock, then pair the Mac and iPhone, then open BleUnlock, the Bluetooth name is displayed correctly! But my Mac device bar shows two Iphone J, I don't know what this is!
[image]
[image]
[image]

ts1, 2020-09-19:
Thanks for the report.
I don't know why two devices are shown. Maybe it is safe to remove one (or both).
Anyways, I'll re-think the way to find the device.

ts1, 2020-10-15:
At version 1.9.0, it displays Bluetooth name and MAC address if available.
I believe this fixed the issue.
Feel free to reopen if you have any problem

closed 2020-10-15 by ts1