ts1/BLEUnlock #21: [Feature request]Enable screen saver when automatically locked

FengYuu, 2020-10-16:
Can you add an option to enable the screen saver when the RSSI lock is triggered instead of directly locking it?

It can be set in Security & Privacy-General to request a password **immediately** after entering the screen saver

When the RSSI unlock is triggered, the password is automatically entered and the screen saver is turned off

Is it possible to achieve? Thank you for your hard work!

ts1, 2020-10-16:
Thanks for the input, @FengYuu!
I'll add that feature when I have time.

ts1, 2020-11-03:
This feature is released in [1.10.0](https://github.com/ts1/BLEUnlock/releases/tag/1.10.0).

closed 2020-11-03 by ts1

FengYuu, 2020-11-03:
Thank you for your hard work! 

But I found a small problem: whenever I trigger the unlock to close the lock screen, the Chinese/English input of the input method will be switched once. 

This input method uses shift to switch this option. I think you used this button? Can other buttons be used?

ts1, 2020-11-04:
Right, I use shift key to show login screen when screensaver is running.
I never thought the shift key is used to switch IM!