ts1/BLEUnlock #24: Can not find Huawei phone

L1cardo, 2020-11-03:
My phone is huawei mate 40 pro, can BLEUnclock can not find my phone.
Even if it can detect, the id of blue tooth is not correct, and the signal will not change after a couple of seconds.

[image]

XuBoZhangQvQ, 2020-11-03:
iPhone 5s or later, Apple Watch (all), or other BLE device that transmits signal periodically
iPhone 5s或更高版本，Apple Watch（全部）或其他定期传输信号的BLE设备
我使用mate40 pro 也发现这个问题了，不能识别手机的名称，而且距离不会发生变化
大概是因为华为手机的蓝牙并没有主动的一直发射信号

L1cardo, 2020-11-03:
> iPhone 5s or later, Apple Watch (all), or other BLE device that transmits signal periodically
> iPhone 5s或更高版本，Apple Watch（全部）或其他定期传输信号的BLE设备
> 我使用mate40 pro 也发现这个问题了，不能识别手机的名称，而且距离不会发生变化
> 大概是因为华为手机的蓝牙并没有主动的一直发射信号

不知道是华为设备没有定期发射信号，还是软件没有检测到（可能频段不同？）， 反正反馈一下。

XuBoZhangQvQ, 2020-11-03:
> > iPhone 5s或更高
> > 版本，Apple Watch（全部）或定期发送信号的BLE设备iPhone 5s或更高版本，Apple Watch（全部）或其他定期传输信号的BLE设备
> > 我使用mate40 pro也发现这个问题了，不能识别手机的名称，而且
> > 距离不会发生变化大概是因为华为手机的蓝牙并没有主动的一直发射信号
> 
> 不知道是华为设备没有定期发射信号，还是软件没有检测到（可能波长不同？），反正反馈一下。

你可以测试一下，打开蓝牙偏好设置，然后打开mate40Pro的蓝牙，这个软件会识别出来一个蓝牙地址，过几秒之后距离就不更新了，这时候关闭蓝牙，打开蓝牙，会变成另外一个地址，每次打开关闭蓝牙都会变为一个新的地址，而且手机的蓝牙信号，不是一直在发射的（大概是为了省电），所以我推断华为的手机（或者安卓设备）都不会主动的实时一直发射信号。而这个软件需要一直检测是否有这个信号，以及他的距离

L1cardo, 2020-11-03:
@XuBoZhangQvQ 当你关闭这个软件以后，Mac搜索到的 mate40 的蓝牙地址是不变的，且为手机的真实蓝牙地址。
但是你打开这个软件以后，mac 能搜索到好几个 mate40 设备，而且蓝牙地址每一次都是不一样的。
所以我猜测和软件也有一定的关系

XuBoZhangQvQ, 2020-11-03:
> @XuBoZhangQvQ 当你关闭这个软件以后，Mac搜索到的 mate40 的蓝牙地址是不变的，且为手机的真实蓝牙地址。
> 但是你打开这个软件以后，mac 能搜索到好几个 mate40 设备，而且蓝牙地址每一次都是不一样的。
> 所以我猜测和软件也有一定的关系

等作者修复吧，我也是用的mate40pro（狗头）

ts1, 2020-11-04:
Thanks for the report!

The BLE MAC address is a complex beast, it can be resolvable random address or unresolvable random address.
Maybe BLEUnlock is displaying the random address.
That behavior is obviously different from Apple devices, and I don’t think I can fix it soon, since I don’t have an Android device with BLE handy at the moment.

However, if you can select the phone, and the No-Signal Timeout is set appropriately, BLEUnlock should work with that phone, regardless of the wrong MAC address.

L1cardo, 2020-11-04:
One more question:
What do `Passive Mode` and `No-Signal Timeout` exactly mean?

ts1, 2020-11-04:
Passive mode is described here: https://github.com/ts1/BLEUnlock#passive-mode-and-bluetooth-internet-sharing
No-signal timeout is time between the last signal reception and unlocking.

L1cardo, 2020-11-04:
> Passive mode is described here: [ts1/BLEUnlock#passive-mode-and-bluetooth-internet-sharing](https://github.com/ts1/BLEUnlock#passive-mode-and-bluetooth-internet-sharing)
> No-signal timeout is time between the last signal reception and unlocking.

Thanks!

ts1, 2020-11-04:
Correction: unlocking → locking

ts1, 2021-02-18:
@L1cardo Is your Huawei phone working with BLEUnlock?
If its MAC address displayed by BLEUnlock changes from time to time, unfortunately it can't be supported.
(See #28)
Anyway I have to update README.

L1cardo, 2021-02-18:
@ts1 emmm sorry that I use iPhone now, can't give you any further feedback.
Apple eco-system pulled me back.

ts1, 2021-02-19:
OK, thanks for the reply.

closed 2021-02-19 by ts1