ts1/BLEUnlock #27: add BLEUnlock to [home]brew?

maelcum, 2021-01-21:
Hello.
This is not an issue, but a feature request. It would be awesome if BLEUnlock would be available via `brew install --cask bleunlock` in brew.sh.
Thank you for your attention.

ts1, 2021-01-22:
Thanks for the input.
I do use Homebrew, but I don't know how it works, how to submit, etc.
It would certainly be nice to be able to install with brew.

nkean, 2021-01-22:
They've got some good documentation for getting setup to contribute.
https://github.com/Homebrew/homebrew-cask/blob/master/doc/development/adding_a_cask.md

joeyhoer, 2021-02-15:
Just made [a PR to homebrew-cask](https://github.com/Homebrew/homebrew-cask/pull/99912) for this.

ts1, 2021-02-16:
@joeyhoer Thank you!

joeyhoer, 2021-02-16:
This issue can be closed, as the PR was accepted into homebrew. 😄

ts1, 2021-02-16:
Confirmed!
Thank you very much @joeyhoer!

closed 2021-02-16 by ts1

maelcum, 2021-02-16:
Thank you all for making this happen!!