ts1/BLEUnlock #28: Save selected device

whitelodge, 2021-02-10:
This is a feature request.

Problem:
After scanning for devices and selecting one to act as the proximity lock/unlock device, the device is not saved as a default that is always expected. After some time of the proximity unlock device being away and connected to other BLE devices, returning back to the comptuer with BLEUnlock installed on it results in no proximity login because BLEUnlock has 'forgotten' the selected device. The device holder then needs to re-scan and find the device again to set it as the proximity unlock device.

Possible Solution:
Being able to save the selected device (with mac address) as a default device that doesn't get forgotten by BLEUnlock. This is only possible if the bluetooth mac address or name doesn't change.

ts1, 2021-02-12:
Usually, the UUID of the device is stored (not the MAC address for some complicated reason).
In recent versions of macOS, the UUID of an Apple device changes every 15 minutes or so, unless you are logged in with the same iCloud ID as your Mac.
If that is the reason, logging in your BLE device and Mac with the same iCloud ID will solve the problem.

whitelodge, 2021-02-12:
I am using an Android device with a Macbook Pro (Big Sur). I'm guessing that the Android with BLE would also randomise the UUID.

ts1, 2021-02-12:
Does BLEUnlock display the correct MAC address of your Android device?
(I'm asking this because BLE device can use **random** MAC address.)
If so, the problem may be solved by modifying BLEUnlock to save MAC address.

whitelodge, 2021-02-12:
BLEUnlock displays the public mac address as a random one and not the bluetooth hardware mac address. This is normal of course. I think that the only way to get an anticipated "ID" between the android (or any phone) and a Mac computer is to pair them. Is that correct?

ts1, 2021-02-13:
Is that random address fixed?
On Apple devices, the random address changes every 15 minutes, and the UUID changes accordingly.
Only if the device uses the same Apple ID as the Mac, the hardware MAC address is resolved in some mysterious way.

On your Android device, the random MAC address doesn't change but the UUID is changing, is that correct?

whitelodge, 2021-02-14:
Unfortunately the public facing mac address changes. However I'm unsure about the UUID changing. I may have to debug to find out what is happening there. In the list of devices, the device name always stays the same. It would be good to have the login occur just by the device name alone, but this poses quite a security hole as it allows for easy spoofing. I'm interested in this project due to it's usefulness and am willing to have it work politely with my google pixel 3a.

ts1, 2021-02-14:
Thank you for the information.
I don't think your device can be supported.
I'll add a note about it in README.

closed 2021-02-14 by ts1