ts1/BLEUnlock #30: BLEU does not list my iPhone by it's BlueTooth

cyberfunk, 2021-04-18:
I can't be sure what device I am linking to BLEU. There are two devices that show up but I think one device might be from a neighbor and the other BlueTooth address does not match the address as reported in my iPhone settings. How can I verify that it's my actual phone that I'm connecting to BLEU? I tried to do a walk away test and the screen did not lock at all with regards to my iPhone's proximity. This just seems a bit vague and I wish I had more certainty about linking my phone e.g. the BlueTooth addresses matching up. Can you shed some light on how to better match a device with BLEU?

ts1, 2021-04-19:
Hello,
Do you sign in with the same Apple ID on Mac and iPhone?
If you use latest OS and sign in with same Apple ID on both Mac and iPhone, the correct MAC address should be acquired.
OTOH, to identify the device, you can watch the dBm value and move the phone away and back, the dBm value should change accordingly.

cyberfunk, 2021-04-19:
Thanks so much for your input and suggestions. In the end I just could not get BLEU to work consistently. Once I did attached the correct device it locked and unlocked the display once successfully. After that I could not get the device to wake my display. It's just not reliable enough to use an I end up having to disable BLEU and revert to the simple built in screen lock. I appreciate your help in sorting out my issues.

closed 2021-04-20 by ts1