ts1/BLEUnlock #35: Disable BLEunlock lock, not system lock

acrostich, 2021-05-10:
On a few occasions when watching things on my laptop (Macbook 12"), but not interacting with it, BLEunlock locked the laptop stating the BLE signal was lost. I then set Lock RSSI to Disable, however this appears to disable the macOS automatic lock too. It would be helpful if it disabled only the BLEunlock one, or an option to enable that.

ts1, 2021-05-11:
BLEUnlock does not prevent system idle sleep.
You might have some process that preventing sleep.
The command below could help find the process.
```
pmset -g
```

acrostich, 2021-05-11:
Thank you. Currently this indicates 'sleep prevented by sharingd'. This disappears if I switch Bluetooth off. I'll need to try to figure out whether that's an issue and if so what is causing this. It's still shows up if I quit BLEunlock so that's unlikely to be the cause.

closed 2021-05-11 by ts1