ts1/BLEUnlock #36: Create a log

acrostich, 2021-05-10:
It would be helpful to have a log of the the fleeting notifications that BLEunlock occasionally display in case of issues with the signal. It would also be helpful, to narrow down any issue, to log the lock and unlock events.

ts1, 2021-05-11:
That can be done with the script.
Copy and paste the commands below in the Terminal.app.
```
cat > ~/Library/Application\ Scripts/jp.sone.BLEUnlock/event << EOF
#!/bin/bash
echo \$(date) \$1 \$2 >>/tmp/bleunlock.log
EOF
chmod +x ~/Library/Application\ Scripts/jp.sone.BLEUnlock/event 
```
The log will be left in `/tmp/bleunlock.log`.

acrostich, 2021-05-13:
You may want to consider as a new feature to have in the UI an option to display, say, the last 10 events. This log is really helpful to understand how BLEunlock works. (I'm currently running a tail -f on it in a terminal window.)