ts1/BLEUnlock #39: RSSI

acrostich, 2021-05-15:
I've switched to passive mode and the reported RSSI seems not as good as when in active mode. I've adjusted the lock and unlock threshold accordingly. I currently have an Unlock RSSI of -60 dBm.

However I noticed in the event log a few unlock events with a stated worse RSSI, such as 'unlocked -66'.

So there are several RSSI used in BLEunlock:
* The instantaneous one shown in the Device list
* The averaged one shown in the menu
* The one used for comparison for locking events
* The one used for comparison for unlocking events
* The one reported for the different types of events in the log

I assumed the last three would be the instantaneous one, but as in some instances the one in the log is not compatible with the one used in comparison for unlocking, either there's another RSSI value or there's a bug in how it's reported in the interface or the log.

I didn't notice such an issue when I had it set to active mode.

(BLEunlock is otherwise working well so far since I switched to passive mode.)

ts1, 2021-05-17:
Instantaneous RSSI is used for unlocking, to make the unlocking more responsive.
The other values are moving average, except for device list.

acrostich, 2021-05-17:
Then I don't understand how with Unlock RSSI set at -60 dBm I have events such as:

unlocked -65
unlocked -70
unlocked -66

I do also have many events that make sense such as:

unlocked -57
unlocked -43
unlocked -59

It otherwise works well.

ts1, 2021-05-17:
All RSSI values in the log are moving averages.

acrostich, 2021-05-17:
Ok, thank you for this further explanation. I now understand it, however I think it would be better to record the instantaneous RSSI in the event log.

The RSSI in the event log is associated with events such as unlock and as you explained the instantaneous RSSI is used for unlocking (and I appreciate the increased responsiveness). So it would be much clearer to record in the event log the specific value that triggered the unlocking.

ts1, 2021-05-18:
I totally agree with your point.
However, please understand that the logging feature is not official, it is just a user scripting feature with RSSI (moving average) added on request.
Having said that, as you mentioned before, it would be better to implement a proper logging feature.
I'll put that in my TODO list.

acrostich, 2021-05-18:
I fully understand and apologise if you felt I was complaining as that was not the case. I agree that adding a logging feature is likely best way forward and possibly also documenting which RSSI is a moving average and which is instantaneous. I am very appreciative of all your work on BLEunlock and the outcome of that work, and my posts here are suggestions for you to consider to possibly making it even better.