ts1/BLEUnlock #44: not work on Monsterey

Teacher-c, 2021-06-09:
can not scan the BLE device

Skyearn, 2021-06-09:
Same problem

Skyearn, 2021-06-09:
Same problem

ts1, 2021-06-10:
I installed Monterey Beta, and reproduced the problem.
And found that if I follow the [Apple's recommendation](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipherals), it can find my SwitchBots, but no Apple devices.
Apple devices don't seem to advertise service UUIDs, other than Exposure Notification, which uses temporary random MAC address so it can't be used for tracking.

Bluetooth Explorer can't find any BLE or Near By devices too.
And this tool is no longer distributed since Xcode 12.

I guess this change in macOS is to enhance privacy.
It might be the right way to go, but not good for BLEUnlock.

The only way to deal with this seems to be to develop apps for iOS, watchOS and maybe Android that advertise BLE beacons in the background.
It remains to be seen how stable background BLE packet transmission will be in these operating systems.
It will need cryptography to implement sufficient anti-tracking measures.
It will be something different from the BLEUnlock of today.

Skyearn, 2021-06-10:
What a pity. BLEUnlock is the best unlock app for me because it doesn't need a iOS or other operating systems app.

Teacher-c, 2021-06-10:
What a pity...
BLEUnlock is the best app to unlock mac without iWatch

ts1, 2021-07-27:
I made some progress with the help of @pjv's information.
If `Application is agent` in Info.plist is set to NO, the device list is now displayed.
Consequently, the BLEUnlock icon will appear in the Dock.

However, there is still a serious problem.
I have not been able to obtain the permanent MAC address (or UUID) of the iPhone or Apple Watch.
Therefore, the connection will be lost in about 20 minutes.
[image]

etrahretep, 2021-07-27:
Strange thing is, I started the app this morning after not working since Monterey beta1. No changes made to the app. It displayed all BLE devices and lock/unlock just works. Even after an hour. Running Monterey beta3 on a late 2015 iMac.
Tried the same on a 2017 MacBook Pro but that doesn't work. No clue why....

etrahretep, 2021-07-27:
And gone in beta4...

Skyearn, 2021-07-28:
> I made some progress with the help of @pjv's information.
> If `Application is agent` in Info.plist is set to NO, the device list is now displayed.
> Consequently, the BLEUnlock icon will appear in the [...]

I clone and build this project get this error: `[open] OSStatus _LSCopyApplicationNodeFromOpenState(LSOpenState *): Returning kLSNoExecutableErr because node is a directory but we failed to register with error -10811`
Does this project require Xcode 11 or earlier?

ts1, 2021-07-28:
@Skyearn I'm building on Xcode 12.5.1 on Monterey.
I'm not familiar with that error but I guess Launcher is not built when building BLEUnlock?
Or something with signing? I'm not sure.

Skyearn, 2021-07-28:
> @Skyearn I'm building on Xcode 12.5.1 on Monterey.
> I'm not familiar with that error but I guess Launcher is not built when building BLEUnlock?
> Or something with signing? I'm not sure.

Should  Launcher be build before build bleulock?

Skyearn, 2021-07-28:
I found that Launcher reported this error.
[image]

ts1, 2021-07-28:
@Skyearn Sorry but I have no idea. I successfully built BLEUnlock on clean installed Monterey and Xcode right out of the box.

Skyearn, 2021-07-29:
@ts1 ok, thanks.

Skyearn, 2021-07-29:
I remove this folder and git clone to build, it’s build success. It's strange that I did the same before, but I reported error 10811. Expect it to run perfectly in macOS 12.

etrahretep, 2021-07-29:
Recompiled for MacOs 12 with Xcode 13b3. UIElement = NO. Working for more then 6 hours now on both machines!

ts1, 2021-07-29:
I've found that if the device is paired, BLEUnlock gets permanent UUID.
I'll need to fix displaying device name and MAC address (Monterey uses SQLite DB instead of plist) before release it.

ts1, 2021-07-29:
Published pre-release that works on Monterey.
https://github.com/ts1/BLEUnlock/releases/tag/1.11-alpha

Skyearn, 2021-07-30:
> Published pre-release that works on Monterey.
> https://github.com/ts1/BLEUnlock/releases/tag/1.11-alpha
It works well for me.

nightwallaby, 2021-07-30:
Pre-release working for me as well

ts1, 2021-08-30:
The first release candidate with Monterey support is now available.
Please test it out and let me know if there are any problems so I can make it a public release.
This should also work on Big Sur and earlier.
https://github.com/ts1/BLEUnlock/releases/tag/1.11-rc

ts1, 2021-09-04:
Fixed bug #53 in [rc2](https://github.com/ts1/BLEUnlock/releases/tag/1.11-rc2). Test it please!

ts1, 2021-09-12:
There were 150 downloads since rc2 was released, and no problems are reported.
So I release it as [1.11](https://github.com/ts1/BLEUnlock/releases/tag/1.11).
Thank you to everyone who helped with development!

closed 2021-09-12 by ts1