ts1/BLEUnlock #45: Failed to get Bluetooth device list

wqq578, 2021-07-20:


ts1, 2021-07-21:
Please let me know the details such as OS version.

wqq578, 2021-07-23:
> Please let me know the details such as OS version.

10.15.7
The first installation will do.
Re-install do not know what the authorization, it seems to point the wrong authorization
After that, the Bluetooth list is not displayed

ts1, 2021-07-27:
Is there Bluetooth in System Preferences > Security & Privacy > Privacy?
(I'm not sure if Catalina has it)
If it is there, make sure BLEUnlock is turned on.

wqq578, 2021-07-27:
> Is there Bluetooth in System Preferences > Security & Privacy > Privacy?
> (I'm not sure if Catalina has it)
> If it is there, make sure BLEUnlock is turned on.

Is there Bluetooth in System Preferences > Security & Privacy > Privacy?
NO

wqq578, 2021-07-27:
[image]

wqq578, 2021-07-27:
[image]

ts1, 2021-07-27:
Sorry, but I don't know what's going on.
Please try the following:
1. Reset the Bluetooth module (Option+Shift+click on the Bluetooth icon in the menu bar).
2. Reset SMC.
3. Reset NVRAM.
4. Remove `/Library/Preferences/com.apple.Bluetooth.plist` in recovery mode (pairing information etc. may be lost).

nigolarer, 2021-11-23:
i ve facing the same issue after i reboot my mac(M1). i ve already permitted Bluetooh to BLUnlock and assistant.

wqq578, 2021-11-23:
> 

> i ve facing the same issue after i reboot my mac(M1). i ve already permitted Bluetooh to BLUnlock and assistant.

wqq578, 2021-11-23:
> i ve facing the same issue after i reboot my mac(M1). i ve already permitted Bluetooh to BLUnlock and assistant.

It's been settled
Two schemes were tried
1 brew Uninstalling software
2 Use other cleaning software to completely uninstall and reinstall

closed 2021-11-23 by wqq578

nigolarer, 2021-11-23:
> 

> > i ve facing the same issue after i reboot my mac(M1). i ve already permitted Bluetooh to BLUnlock and assistant.
> 
> It's been settled Two schemes were tried 1 brew Uninstalling software 2 Use other cleaning software to completely uninstall and reinstall

Thx for ur reply, but after un&reinstall and clean up by some tmp file cleaner app, this problem still unresolved. Could u plz offer config or tmp file directory that this app will write to? Thx a lot, this app really give me such a big convenience.

wqq578, 2021-11-23:
> Thx for ur reply, but after un&reinstall and clean up by some tmp file cleaner app, this problem still unresolved. Could u plz offer config or tmp file directory that this app will write to? Thx a [...]

You can add my contact information wqq578. I used the third-party software to uninstall it

wqq578, 2021-11-23:
I remember uninstalling it using ‘CleanMyMac X’, the free version. Uninstall finished, I also cleaned up the software

nigolarer, 2021-11-24:
> I remember uninstalling it using ‘CleanMyMac X’, the free version. Uninstall finished, I also cleaned up the software

thx a lot, this clean app literally worked and BLEUnlock on my mac well as before now :)
FYI: Though `CleanMyMac X` free version have soooo  many limits, it shows app and related file path clearly and can easy to remove manually by oneself.