ts1/BLEUnlock #46: Support iBeacon

pjv, 2021-07-21:
I don’t know if this is a potential solution to #44, but it would be a solution to android phones not advertising a static mac address. There are any number of apps for Android that let you set them to transmit a BLE iBeacon advertisement. If BLEUnlock had a preference to allow it to scan for iBeacons, it would be able to support use with modern Android devices running such an app.

Here is a project that has macOS code for scanning for iBeacons - the release of the demo app from this project reliably identifies my Android iBeacon:

https://github.com/mlwelles/BeaconScanner

ts1, 2021-07-22:
Thanks for the information.
I have an iBeacon (real thing) and BLEUnlock supports it and shows the approximate distance.
[image]

And, I tried the BeaconScanner on Monterey and it works!
I wonder why, because the two projects do the basically same thing. (CBCentralManager.scanForPeripherals)
Anyway, there is hope that we can support Monterey.
Thanks again!

etrahretep, 2021-07-25:
Oh yes please! Missing it daily!

ts1, 2021-07-27:
I close this issue because iBeacon is already supported.
Thanks to @pjv for the information about BeaconScanner, Monterey support has progressed a bit (not yet completed though).
Discussion about Monterey will be continued in #44.

closed 2021-07-27 by ts1