ts1/BLEUnlock #47: Keep getting flashbacks

sanwan1174, 2021-07-30:
BLEUnlock version: 1.10.2 (735), I also tried the latest pre-released version, its still not work.
System version: macOS 11.4
CPU: Apple Silicon M1

Skyearn, 2021-07-30:
https://github.com/ts1/BLEUnlock/issues/44#issuecomment-888971305

ts1, 2021-07-30:
What exactly does flashback mean?

sanwan1174, 2021-07-30:
App crash.

sanwan1174, 2021-07-30:
This condition occurs when it scanning for Bluetooth devices.

ts1, 2021-07-30:
Could you send me the detail of crash report?
It will look like below
```
Process:               BLEUnlock [5199]
Path:                  /Users/USER/Desktop/BLEUnlock.app/Contents/MacOS/BLEUnlock
Identifier:            jp.sone.BLEUnlock
Version:               1.11-alpha (746)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           BLEUnlock [5199]
User ID:               501

Date/Time:             2021-07-30 14:31:28.618 +0900
OS Version:            macOS 11.5.1 (20G80)
Report Version:        12
Anonymous UUID:        C7AC3BFF-A28E-4727-2566-ECF852F41761

Sleep/Wake UUID:       0B5EA535-47B1-48D1-8268-556B9BDC3684
...
```

sanwan1174, 2021-07-30:
Process:               BLEUnlock [1548]
Path:                  /Applications/BLEUnlock.app/Contents/MacOS/BLEUnlock
Identifier:            jp.sone.BLEUnlock
Version:               1.11-alpha (744)
Code Type:             ARM-64 (Native)
Parent Process:        ??? [1]
Responsible:           BLEUnlock [1548]
User ID:               501

Date/Time:             2021-07-30 14:57:22.402 +0800
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Anonymous UUID:        454E48D2-4D6B-D17C-E633-BABDAF9E5276

Time Awake Since Boot: 420 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       EXC_ARM_BREAKPOINT at 0x0000000104e96144 (brk 1)
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Trace/BPT trap: 5
Termination Reason:    Namespace SIGNAL, Code 0x5
Terminating Process:   exc handler [1548]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   jp.sone.BLEUnlock             	0x0000000104e96144 0x104e88000 + 57668
1   jp.sone.BLEUnlock             	0x0000000104e95d80 0x104e88000 + 56704
2   jp.sone.BLEUnlock             	0x0000000104e95784 0x104e88000 + 55172
3   jp.sone.BLEUnlock             	0x0000000104e9323c 0x104e88000 + 45628
4   jp.sone.BLEUnlock             	0x0000000104e8c0cc 0x104e88000 + 16588
5   jp.sone.BLEUnlock             	0x0000000104e98754 0x104e88000 + 67412
6   jp.sone.BLEUnlock             	0x0000000104e989ec 0x104e88000 + 68076
7   com.apple.CoreBluetooth       	0x00000001b19a0edc -[CBCentralManager handlePeripheralDiscovered:] + 1448
8   com.apple.CoreBluetooth       	0x00000001b19a2c90 -[CBCentralManager handleMsg:args:] + 288
9   com.apple.CoreBluetooth       	0x00000001b1994b5c -[CBManager xpcConnectionDidReceiveMsg:args:] + 208
10  com.apple.CoreBluetooth       	0x00000001b1994a6c __30-[CBXpcConnection _handleMsg:]_block_invoke + 68
11  libdispatch.dylib             	0x00000001a1972128 _dispatch_call_block_and_release + 32
12  libdispatch.dylib             	0x00000001a1973ec0 _dispatch_client_callout + 20
13  libdispatch.dylib             	0x00000001a197b6a8 _dispatch_lane_serial_drain + 620
14  libdispatch.dylib             	0x00000001a197c2d8 _dispatch_lane_invoke + 456
15  libdispatch.dylib             	0x00000001a1982200 _dispatch_main_queue_callback_4CF + 608
16  com.apple.CoreFoundation      	0x00000001a1c62d28 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 16
17  com.apple.CoreFoundation      	0x00000001a1c2173c __CFRunLoopRun + 2524
18  com.apple.CoreFoundation      	0x00000001a1c205e8 CFRunLoopRunSpecific + 600
19  com.apple.HIToolbox           	0x00000001a9b3b2a0 RunCurrentEventLoopInMode + 292
20  com.apple.HIToolbox           	0x00000001a9b3b014 ReceiveNextEventCommon + 552
21  com.apple.HIToolbox           	0x00000001a9b3add4 _BlockUntilNextEventMatchingListInModeWithFilter + 72
22  com.apple.AppKit              	0x00000001a440f480 _DPSNextEvent + 836
23  com.apple.AppKit              	0x00000001a440de20 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1292
24  com.apple.AppKit              	0x00000001a4b86f70 _NSHLTBMenuEventProc + 340
25  com.apple.HIToolbox           	0x00000001a9ba39e4 IsUserStillTracking(MenuSelectData*, unsigned char*) + 216
26  com.apple.HIToolbox           	0x00000001a9bb8330 CheckDragging(MenuSelectData*, unsigned char*) + 184
27  com.apple.HIToolbox           	0x00000001a9ccd6f0 TrackMenuCommon(MenuSelectData&, unsigned char*, SelectionData*, MenuResult*, MenuResult*) + 1596
28  com.apple.HIToolbox           	0x00000001a9cc9d40 PopUpMenuSelectCore(MenuData*, Point, double, Point, unsigned short, unsigned int, unsigned int, Rect const*, unsigned short, unsigned int, Rect const*, Rect const*, __CFDictionary const*, __CFString const*, OpaqueMenuRef**, unsigned short*) + 1152
29  com.apple.HIToolbox           	0x00000001a9cc94a4 _HandlePopUpMenuSelection8(OpaqueMenuRef*, OpaqueEventRef*, unsigned int, Point, unsigned short, unsigned int, unsigned int, Rect const*, unsigned short, Rect const*, Rect const*, __CFDictionary const*, __CFString const*, OpaqueMenuRef**, unsigned short*) + 600
30  com.apple.HIToolbox           	0x00000001a9b91bd0 _HandlePopUpMenuSelectionWithDictionary + 380
31  com.apple.AppKit              	0x00000001a48cb4fc SLMPerformPopUpCarbonMenu + 1796
32  com.apple.AppKit              	0x00000001a4769344 _NSSLMPopUpCarbonMenu3 + 808
33  com.apple.AppKit              	0x00000001a4768f2c -[NSCarbonMenuImpl popUpMenu:atLocation:width:forView:withSelectedItem:withFont:withFlags:withOptions:] + 392
34  com.apple.AppKit              	0x00000001a4b794dc +[NSStatusBarButtonCell popupStatusBarMenu:ofBar:inRect:ofView:withEvent:] + 400
35  com.apple.AppKit              	0x00000001a4b78824 -[NSStatusBarButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 192
36  com.apple.AppKit              	0x00000001a462332c -[NSControl mouseDown:] + 636
37  com.apple.AppKit              	0x00000001a46216b4 -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 4384
38  com.apple.AppKit              	0x00000001a4592064 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2444
39  com.apple.AppKit              	0x00000001a459146c -[NSWindow(NSEventRouting) sendEvent:] + 352
40  com.apple.AppKit              	0x00000001a4b7af5c -[NSStatusBarWindow sendEvent:] + 836
[... 436 lines cut ...]
[... cut ...]
AirPort: spairport_wireless_card_type_airport_extreme, wl0: Apr 28 2021 19:53:33 version 18.50.40.7.7.8.119 FWID 01-e6894c69
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: USB 10/100/1000 LAN, Ethernet, en6
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: USB 2.0 BILLBOARD
USB Device: USB Storage
USB Device: USB 3.1 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

ts1, 2021-07-30:
@sanwan1174  Could you join my Discord? So I will share a special build to identify the problem.
https://discord.gg/QbTJqEBSfR

levie-vans, 2021-08-12:
> I also meet the same problem.When I get a list of Bluetooth devices, the app flashes.
> 
> MacBook Pro (15-inch, 2018)
> OS：macOS Big Sur 11.5 (20G71)

Process:               BLEUnlock [23831]
Path:                  /Applications/BLEUnlock.app/Contents/MacOS/BLEUnlock
Identifier:            jp.sone.BLEUnlock
Version:               1.11-alpha (744)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           BLEUnlock [23831]
User ID:               501

Date/Time:             2021-08-12 14:16:02.536 +0800
OS Version:            macOS 11.5 (20G71)
Report Version:        12
Bridge OS Version:     5.5 (18P4759a)
Anonymous UUID:        E5E57535-9E51-60BC-6361-CE5D10FAB225

Sleep/Wake UUID:       BA9829CA-F300-4852-A1E9-7C8BDA53143F

Time Awake Since Boot: 250000 seconds
Time Since Wake:       74000 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [23831]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   jp.sone.BLEUnlock             	0x000000010a89be28 0x10a88f000 + 52776
1   jp.sone.BLEUnlock             	0x000000010a89b3bc 0x10a88f000 + 50108
2   jp.sone.BLEUnlock             	0x000000010a898a59 0x10a88f000 + 39513
3   jp.sone.BLEUnlock             	0x000000010a8915f2 0x10a88f000 + 9714
4   jp.sone.BLEUnlock             	0x000000010a89e5dd 0x10a88f000 + 62941
5   jp.sone.BLEUnlock             	0x000000010a89e7ab 0x10a88f000 + 63403
6   com.apple.CoreBluetooth       	0x00007fff30a3f271 -[CBCentralManager handlePeripheralDiscovered:] + 1703
7   com.apple.CoreBluetooth       	0x00007fff30a41048 -[CBCentralManager handleMsg:args:] + 811
8   com.apple.CoreBluetooth       	0x00007fff30a33037 -[CBManager xpcConnectionDidReceiveMsg:args:] + 177
9   com.apple.CoreBluetooth       	0x00007fff30a32f73 __30-[CBXpcConnection _handleMsg:]_block_invoke + 66
10  libdispatch.dylib             	0x00007fff20310623 _dispatch_call_block_and_release + 12
11  libdispatch.dylib             	0x00007fff20311806 _dispatch_client_callout + 8
12  libdispatch.dylib             	0x00007fff203175ea _dispatch_lane_serial_drain + 606
13  libdispatch.dylib             	0x00007fff203180e0 _dispatch_lane_invoke + 417
14  libdispatch.dylib             	0x00007fff2031da08 _dispatch_main_queue_callback_4CF + 613
15  com.apple.CoreFoundation      	0x00007fff205f15a8 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
16  com.apple.CoreFoundation      	0x00007fff205b37a2 __CFRunLoopRun + 2755
17  com.apple.CoreFoundation      	0x00007fff205b261c CFRunLoopRunSpecific + 563
18  com.apple.HIToolbox           	0x00007fff287f7a83 RunCurrentEventLoopInMode + 292
19  com.apple.HIToolbox           	0x00007fff287f77e5 ReceiveNextEventCommon + 587
20  com.apple.HIToolbox           	0x00007fff287f7583 _BlockUntilNextEventMatchingListInModeWithFilter + 70
21  com.apple.AppKit              	0x00007fff22db9502 _DPSNextEvent + 864
22  com.apple.AppKit              	0x00007fff22db7cd5 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
23  com.apple.AppKit              	0x00007fff23534472 _NSHLTBMenuEventProc + 213
24  com.apple.HIToolbox           	0x00007fff28860047 IsUserStillTracking(MenuSelectData*, unsigned char*) + 195
25  com.apple.HIToolbox           	0x00007fff28873f14 CheckDragging(MenuSelectData*, unsigned char*) + 219
26  com.apple.HIToolbox           	0x00007fff2897cba8 TrackMenuCommon(MenuSelectData&, unsigned char*, SelectionData*, MenuResult*, MenuResult*) + 1730
27  com.apple.HIToolbox           	0x00007fff28978f3b PopUpMenuSelectCore(MenuData*, Point, double, Point, unsigned short, unsigned int, unsigned int, Rect const*, unsigned short, unsigned int, Rect const*, Rect const*, __CFDictionary const*, __CFString const*, OpaqueMenuRef**, unsigned short*) + 1274
28  com.apple.HIToolbox           	0x00007fff2897854a _HandlePopUpMenuSelection8(OpaqueMenuRef*, OpaqueEventRef*, unsigned int, Point, unsigned short, unsigned int, unsigned int, Rect const*, unsigned short, Rect const*, Rect const*, __CFDictionary const*, __CFString const*, OpaqueMenuRef**, unsigned short*) + 410
29  com.apple.HIToolbox           	0x00007fff2884d019 _HandlePopUpMenuSelectionWithDictionary + 329
30  com.apple.AppKit              	0x00007fff23286d7a SLMPerformPopUpCarbonMenu + 2227
31  com.apple.AppKit              	0x00007fff231226f1 _NSSLMPopUpCarbonMenu3 + 1096
[... 460 lines cut ...]
Memory Module: BANK 0/ChannelA-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E1
Memory Module: BANK 2/ChannelB-DIMM0, 8 GB, DDR4, 2400 MHz, Micron, 8ATF1G64HZ-2G6E1
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Jul  7 2021 02:34:10 version 9.30.444.17.32.5.70 FWID 01-4075fabc
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: USB 10/100/1000 LAN, Ethernet, en12
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Miscellaneous Device
USB Device: Gaming Keyboard G610
USB Device: USB to PCIE Bridge
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB3.0 Hub
USB Device: Apple T2 Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5

ts1, 2021-08-12:
I appreciate your crash dumps, but in the future please use https://pastebin.com/ or something, I'm tired of scrolling.

ts1, 2021-08-12:
@sanwan1174 @levie-vans 
Please try this build to see if it fixes crashes:
https://github.com/ts1/BLEUnlock/releases/tag/1.10.3-rc4

levie-vans, 2021-08-12:
> @sanwan1174 @levie-vans
> Please try this build to see if it fixes crashes:
> https://github.com/ts1/BLEUnlock/releases/tag/1.10.3-rc4

The problem is solved, it works well now!  Thanks!

ts1, 2021-08-29:
I believe this problem is fixed in release 1.10.3.
Feel free to reopen if it is not.

closed 2021-08-29 by ts1