ts1/BLEUnlock #53: Settings are greyed out

AlexKusnezov, 2021-09-02:
I'm on Big Sur 11.5.2
When I launch BLEUnlock everything is greyed out, I can't even quit the app.

[image]

ts1, 2021-09-03:
Is it 1.11-rc?
I have seen that too, but when I restarted the app, it never reproduce.
To quit type this command in Terminal.
```
pkill BLEUnlock
```

ts1, 2021-09-03:
I found the cause of the problem and will fix it.
In the mean time, please use [1.10.3](https://github.com/ts1/BLEUnlock/releases/tag/1.10.3) instead of the pre-release version.

AlexKusnezov, 2021-09-03:
@ts1 thanks this one works

commit 2cd34b90 (2021-09-04): Fixed bug that modal disappears on first launch

Closes #53.
closed 2021-09-12 by ts1