ts1/BLEUnlock #57: Saving Energy

cyberclaus, 2021-12-04:
First of all thank you for this great app. Now to my question: do you think it is possible to add "Switch off display"? I guess that would save some energy and make the world a little bit greener.

ts1, 2021-12-06:
Do you mean the option to turn off the screen when BLEUnlock locks the Mac?
I'll do it when I have time.

cyberclaus, 2021-12-06:
Exactly. I'm not good enough in C programming for helping you out. But "pmset displaysleepnow" should do the work.

In addition to that can I deliver localizations (translations) for German and Swedish which I already tested. Could also do some for Danish and Norwegian (Bokmål).

ts1, 2021-12-06:
As for localization, please make a Pull Request and I'll be happy to merge it!