ts1/BLEUnlock #68: During setup, I mistakenly selected my printer, how can I remove my printer then select my phone?

yaoyao082008, 2021-12-18:
During setup, I mistakenly selected my printer, how can I remove my printer then select my phone?

ts1, 2021-12-18:
Open Devices menu again and select your phone. 😄

yaoyao082008, 2021-12-18:
I only the "Device" menu (Not "Devices"), clicking it only shows a grayed-out "Scanning".  Where does BLEUnlock store its config? thx

ts1, 2021-12-18:
Config is at `~/Library/Preferences/jp.sone.BLEUnlock.plist`.
Maybe removing it doesn't help your situation.
Try exit BLEUnlock and launch again.
If it is still 'Scanning...' only,  then restart the Bluetooth system by entering the following command on Terminal then restart BLEUnlock.
```
sudo pkill bluetoothd
```

yaoyao082008, 2021-12-18:
after "sudo pkill bluetoothd" and "reset bluetooth module",  I can see only my printer prefixed with a check mark.   Still can not my phone, which I did see when I ran BLEUnlock for the first time. Thx

ts1, 2021-12-18:
Just wait and see until the phone appears.
Also connect Bluetooth between Mac and phone would help.