ts1/BLEUnlock #76: Can't see any device, only shows scanning. Passive mode didn't help, no reset bluetooth in Monterey

electr1fy0, 2022-01-21:


ts1, 2022-01-21:
Try this on Terminal:
```
sudo pkill bluetoothd
```
It will ask your login password.

I have to write this in README.

electr1fy0, 2022-01-21:
Tried, didn't help.
Do you think Android devices will be discovered by it?

a-little-cat, 2022-01-24:
Same situation.
First installed, I can see all the devices, and I choose my iPhone.
After a few days, the list is always scanning, but my iPhone block and unblock mac normally.
Sudo pkill bluetoothd didn't help.
Is there another way to reset all the config?

a-little-cat, 2022-01-25:
1.12.1 didn't help.
After upgrade BLEUnlock, my iPhone is lost, and the list is scanning.
I uninstall and install BLEUnlock with brew, my iPhone is back, but the list is still scanning.

a-little-cat, 2022-01-25:
Besides, my mac is m1 mbp13 2020, os is 12.1, I hope this will help.

GraveyardEnthuisast, 2022-02-22:
Same. I'm having this same issue. It worked once and then after a while it just said scanning and it never finds it

GraveyardEnthuisast, 2022-02-22:
I have a m1 mac as well

AakLak, 2022-02-22:
Same issue (Devices says "Scanning...") on:
OnePlus 6T on Android 10.14.6
MacOS 10.15 on MacBook Air 2015

Other iOS devices in the area are found.

TIM952597205, 2022-05-07:
Reboot the App, MAC doesn't seem to work, reinstalled 1.12.1, granted Bluetooth authorization, but still scanning forever...

ts1, 2022-05-07:
@TIM952597205 Do you use M1 Mac?

TIM952597205, 2022-05-07:
> @TIM952597205您使用 M1 Mac 吗？
MacBook Pro (13-inch, 2019, Two Thunderbolt 3 ports) Intel Core i5

ts1, 2022-05-07:
Hmm 🤔

TIM952597205, 2022-05-09:
> Hmm 🤔

All of a sudden today I'm fine. I don't know what I did. It's weird

sdonnet, 2022-05-10:
Same thing. No devices list. Just scanning...
"pskill bluetoothd" does nothing.
I need to quit BLEUnlock, and restart BLEUnlock, and I get a devices list, including my phone.
Quite annoying.

I am MacOS Monterey, 12.2, M1.

tinyapps, 2022-05-14:
Same issue on a MacBook Air (Retina, 13-inch, 2020) paired with an iPhone 6s and running BLEUnlock version 1.12.1 (778). Just have to quit and restart BLEUnlock a couple of times to get it working again. Thanks as ever for crafting and sharing this, @ts1!

a-little-cat, 2022-05-15:
@AakLak 
Hello, bro.
May I ask a question about how to solve the android random bluetooth mac address problem?
Thanks very mush.

shisaq, 2022-08-24:
I have the same issue. Have tried the following methods but still keeps scanning:

1. sudo pkill bluetoothd
2. restart the app
3. reinstall the app
4. restart my laptop

So currently it just can't lock my screen when I take my iPhone and leave my laptop. It shows -44dBm EVEN my iPhone's bluetooth is off. When I try to disconnect my device and reselect it, it just keeps scanning forever.

System version: macOS Monterey 12.5.1
Hardware: MacBook Pro 2019 with Intel core

abstarstrong, 2022-11-22:
I'm still having this issue as well. Tried everything.

System version: macOS High Sierra 10.13.6 (17G14042)
Hardware: iMac (21.5-inch, Late 2009) [3.06 GHz Intel Core 2 Duo]

Would love to get this working.
Thanks.

ts1, 2022-11-23:
@abstarstrong iMac Late 2009 doesn't seem to have Bluetooth LE.