ts1/BLEUnlock #77: Can an Android app be made to fix the MAC address randomization problem?

electr1fy0, 2022-01-21:


ts1, 2022-01-21:
Possibly.
However, at the moment I don't have a recent Android device handy to reproduce the problem.

Shockshwat, 2022-02-25:
I do not think so, However, Getting a custom OS might be giving this feature as its not by google

MacsInSpace, 2022-04-15:
Just a suggestion... Can the app read the devices in the default bluetooth plist file?  Either the DeviceCache to get the device name from lister MAC addresses?
/Library/Preferences/com.apple.Bluetooth.plist

That might be a security risk I guess.  Just change my bluetooth device name to the name of the unlock device.

Just thinking out loud

ts1, 2022-04-15:
> /Library/Preferences/com.apple.Bluetooth.plist

BLEUnlock already reads that file to obtain MAC address and device name.
https://github.com/ts1/BLEUnlock/blob/master/BLEUnlock/BLE.swift#L10

MacsInSpace, 2022-07-25:
I just updated to an ARM 16inch MBPro and on this, i can now see the device by name that i couldnt before. 
Pixel 6 Pro (on the Android 13 beta but not paired to my new Mac yet)and other non Apple devices.

It's strange..  I'm wondering if it's a MacOS Monterey thing or an Android 13 Beta thing.

I don't have my google watch (Fossil 5) with me today but will check in a few hours time.

[image]

ts1, 2022-07-25:
It may not work because the MAC address is not displayed.
Might it show up when the phone is paired?

I recently got a cheap Android 11 phone and it did not show up in BLEUnlock even when paired.

MacsInSpace, 2022-07-25:
You're right. It uses it ok to lock/unlock when "out of range" okay but if BT is switched off, it doesn't unlock as it seems to forget the device.
I'm guessing that I am testing the same stuff and getting the same results as you already have in this case. 
Sorry for the mail spam. Just keen to try to help as its a great product. 
As a work around, I've paired it to my Tile which is attached to my keys for now...