ts1/BLEUnlock #82: Anyway to sort the Device list by signal strength (RSSI/dBm)?

heatwave998, 2022-05-19:
I guess this is more of a feature request. Is it possible to have the Devices list be sorted by signal strength? When in busy device environments, it's not easily possible to discern one iphone or apple watch from another, since most devices just show up with generic names in BLE. 

Sorting ability would be awesome!

ts1, 2022-05-20:
Thanks for the input.
I always thought so too, but for implementation reasons it was hard to make the sorting work.
If I can find a better way, I'll try to implement it.

jurnlee, 2022-07-22:
Default sort by name? Keep the devices list for few mimutes maybe, to prevent signal strength changing so quickly. (poor en)