ts1/BLEUnlock #87: [feature request] scheduled active state

solomrdolo, 2022-11-01:
Would be useful if there was an option to set a schedule for when the app is active. For instance I could have the app active during 5:30 am till 12:30pm. After 12:30 the app will be deactivated until the next day.

solomrdolo, 2022-11-02:
Hmm interesting. Let's say the app is deactivated during sleep and the Mac has a scheduled wake/sleep. And both the Mac and crontab (or something) will be scheduled to wake and activate at 5:30 AM. Would it be able to activate with one of those scripts upon Macs scheduled wake? That way it will unlock the mac right when the mac wakes up.

ts1, 2022-11-02:
BLEUnlock can be started/stoped using `open /Applications/BLEUnlock.app` / 'pkill BLEUnlock`.
Mac can be started/stoped by pmset, caffeinate or sometihing.

solomrdolo, 2022-11-02:
I'm for sure going to try this later today. Thank you for the suggestion

solomrdolo, 2022-11-02:
Crontab works like a charm 🤘. Thanks for the tip

closed 2022-11-03 by ts1