ts1/BLEUnlock #90: Macbook using BLEUnlock are at risk of being arbitrarily unlocked

migraine-sudo, 2022-11-04:
# Macbook using BLEUnlock are at risk of being arbitrarily unlocked

## Vulnerability Type :

Hardware/Radio Vulnerability

## Vulnerability Version :

Open source software BLEUnlock https://github.com/ts1/BLEUnlock

Version: 1.12.1 (latest version)

## Recurring environment:

Operating environment: Mac OS (MacBook Pro 13)

## Vulnerability Description AND recurrence:

[image]

For the normal use process of this software, first select the Bluetooth broadcast of our device, for example, choose Xiaomi Mi Band 7 here, and the address is E7:BA:BE:19:79:72(random). It only needs the broadcast signal RSS less than 60db to automatically unlock the computer, because the Bluetooth address is not really unique, which constitutes a security risk.

[image]

According to the principle of the document, bleunlock is judged by the Mac address of BLE. This results in a lack of sufficiently secure authentication for Bluetooth devices. Directly forging a fake bluetooth broadcast can deceive the macbook to unlock. The bluetooth broadcast here is fixed and public, so this is easy. For details, see the video below.

**Video link ---> https://www.bilibili.com/video/BV1LP411A7jv/**

**Attack process**

>Step1: Grab the Bluetooth broadcast address of the Bluetooth device bound to UnlockBLE
>
>Step2: Use the development board to fake bluetooth broadcast (NRF TI whatever..)
>
>Step3: Close to the macbook to unlock, the attacker successfully cracks the macbook.

ts1, 2022-11-04:
Indeed, BLEUnlock is vulnerable to spoofed MAC addresses.
I recommend using BLEUnlock only for locking, disabling Unlock RSSI, and using Touch ID or other secure method for unlocking.
This should be clearly stated in the README.

ts1, 2022-11-04:
I am thinking of using public key cryptography, where the mobile device holds the private key, BLEUnlock on a Mac sends a nonce to the device, the device signs it, and the Mac verifies it with the public key.
Of course, a dedicated application will be needed on the mobile device side.

migraine-sudo, 2022-11-10:
Thanks for your reply, maybe you can try LTK using bluetooth with OOB binding. Index LTK is required to connect before unlocking, but some devices may not support it.

ts1, 2022-11-10:
Thanks for the information. I'll have to learn a lot to implement it.