ts1/BLEUnlock #96: Bluetooch earphones is affected

mianzhiwj, 2023-02-15:
The signal of Bluetooth earphones is affected when the software is running in the background. If a Bluetooth mouse or trackpad has been used, the sound of the Bluetooth earphones may intermittently cut out.

ts1, 2023-02-15:
Could you try enabling Passive Mode?

mianzhiwj, 2023-02-15:
When I turned the software back on today, passive mode was on. I'm not sure if I turned the passive mode on a few months ago when I felt the headphones were affected. is it on by default?

I just tested it, and it did feel like the sound was unaffected in passive mode. 

Thanks！

ts1, 2023-02-15:
It is off by default.

Bluetooth audio is also affected by other things, such as 2.4GHz WiFi.
Note that it is especially likely to be affected when your Mac is connected to 2.4GHz WiFi.

closed 2023-02-15 by ts1