ts1/BLEUnlock #99: [feature suggestion] BLEUnlock icon in the dock

michelescippa, 2023-04-10:
Every time the device is locked via BLEUnlock (that works fine for me!) the icon appears on the dock and I have to manually remove.
It would be nice if it remains only on the appbar.
Thank you

ts1, 2023-04-10:
Hi there,

Thanks for reporting the issue with BLEUnlock. Since this is the first time I've heard about this problem, I'd like to get more information to understand what might be causing it.

Can you please let us know:

Your macOS version
The version of BLEUnlock you're using
To anyone else who might be viewing this issue: if you are experiencing the same problem, please feel free to share your details as well. This info will help me look into any potential compatibility issues. Thanks in advance for your help!

cypok, 2025-03-05:
Hi, I'm experiencing the same problem. The icon appears in the right half of the dock (non-pinned apps). It happens every time after locking, even if I remove it from the dock.

macOS: 15.3.1 (24D70)
BLEUnlock 1.12.2 (795)

[image]