ts1/BLEUnlock #101: My device is not visible

krushnadeore, 2023-07-10:
Poco X4 5g
Mac address is static still my device is not visible in device list 
I tried all options like passive mode and others
even I am able to connect my phone to Mac bluetooth and transfer file still device is not visible in BLEunlock app 
Or is there any way I can add key inside .plist file by converting my phones Mac address ?

tryfunds-admin, 2023-07-11:
My Android (OPPO Reno5 A, Android 12) is not visible either. ☹️
It's visible in System Preferences.
I guess recent Android is not compatible to Apple's CoreBluetooth.
It's not possible to manually insert because it needs UUID, not MAC address.

ts1, 2023-07-11:
My Android (OPPO Reno5 A, Android 12) is not visible either. ☹️
It's visible in System Preferences.
I guess recent Android is not compatible with CoreBluetooth.
It's not possible to add it manually because it needs UUID that is created by CoreBluetooth.

closed 2023-11-11 by krushnadeore