ts1/BLEUnlock #103: Cannot unlock mac on macOS14 Sonoma

lightumcc, 2023-08-24:
It can only display the TouchID, but no password input frame
[image]

TomatoFryEggs, 2023-09-28:
重选下设备就好了

ts1, 2023-09-30:
Hey there! I can't seem to reproduce this bug on my end. If you're experiencing it, could you please provide more details about your environment? It might help narrow things down. Thanks! 😊

QuantumRoseinAmethystVase, 2023-10-25:
This has nothing to do with BLEUnlock. By default, Mac hides the password input box when waking up the screen, and it appears after a single click on the screen.

bren0524, 2024-01-16:
Same problem appeared after I update both my macmini and  macbook air to sonoma,bleunlock cant unlock my sonoma system.

closed 2024-05-22 by lightumcc