ts1/BLEUnlock #106: M1 Macbook Air Won't Auto Unlock

alvinpetitt, 2023-08-31:
Machine is an M1 Macbook Air running macOS 13.5.1 (Ventura). Machine locks as expected but won't automatically unlock. I've uninstalled and reinstalled BLEUnlock but am still getting the same results. Have tried multiple values in the Unlock RSSI to no avail. Device is set to my Apple Watch Ultra.

ts1, 2023-09-07:
Hi, please make sure that you have set your password and the option *Wake without Unlocking* is unchecked.
With this setting, BLEUnlock unlocks my M2 Air running 13.5.1 without a problem.
[image]

alvinpetitt, 2023-10-27:
I'm sorry for this long delay. I was thinking that I would get a notification on replies (obviously have to check settings). Anyway, I've now updated to Sonoma. I'm still having the same issue even after doing what you suggested above. But since it's a different version of macOS, I realize it's irrelevant now. Thank you for the reply.