ts1/BLEUnlock #108: Very rare issue: iPhone 15 Pro Max and AirPods Pro 2 audio connection issue while bleunlock is activated

shakisha, 2023-10-04:
After a lot of investigations I have discovered that randomly, when bleunlock is opened on my MacBook Pro m2, the phonecalls that arrives on my iPhone 15 pro max connected with my AirPods Pro 2, are having got NO AUDIO at all 70% of the time.

To reproduce, if I use Siri using the AirPods button to tell "call someone" then I'll will not have got any audio but the called party will receive the call with mute audio. Very strange issue.

How to investigate or debug this?

ts1, 2023-10-04:
Although the issue is likely on the iOS side, have you tried Passive Mode?

shakisha, 2023-10-05:
tomorrow i’ll try better but in some first trials with the passive mode the issue disappears.

shakisha, 2023-10-05:
I can confirm that in 80 calls done, no issues with passive mode.