ts1/BLEUnlock #113: [feature suggestion] Add option to only unlock with Apple Watch when watch is worn/unlocked, but not when locked

DigiH, 2023-12-15:
Suggestion to add option to only unlock with Apple Watch when watch is worn/unlocked, but not when locked.

Currently my Apple Watch nicely unlocks my iMac when I wear it and enter the room, but it also does the same unlocking when it is locked on the charger, as solely its BT MAC address is evaluated for unlocking.

If additionally also the BLE Apple Watch advertising broadcast data would be evaluated it would allow for an extra security factor to only unlock when the Apple Watch is itself unlocked, worn or on the charger.

The Apple Watch BLE 'Nearby Info' advertising manufacturerdata is 18 hex bytes long and starts with

`4c001005…`

with the octet at index 10 (starting from index 0) either being "98" for **unlocked** or "18" for **locked**.

So a broadcast like 

`4c00100503989c6bf9`

would be unlocked, and 

`4c0010050b182068dd`

locked.

Archetrix, 2024-04-19:
As this would add a step to identify a device beyond just checking for BT MAC and RSSI is there a way to maybe identify that we're looking at an Apple Watch by it's BT MAC already? This way the additional checks would only be executed on an Apple Watch.

On the other hand this check only needs to be done once when the device appears in range (RSSI lower than set threshold) and then only ever again when the RSSI has gone over the threshold.

Thinking further this check should always be made. What if i take off my Apple Watch in front of the Mac to charge it and then it would not check again. In locking my computer by taking off my Apple Watch i'm surely aware of the implications.

ts1, 2024-04-19:
We should be able to identify whether it's an Apple Watch by the device name (Apple Inc./WatchX,X). I'm still considering how it should behave when the Watch is removed from the wrist to charge while the Mac is unlocked.

DigiH, 2024-04-19:
My above suggested Nearby Info Continuity BLE advertising data, looking at the data length and the relevant octet being "98" or "18" clearly identifies it is an Apple Watch, so any name comparison or so is necessary.

https://github.com/theengs/decoder/blob/development/src/devices/APPLEWATCH_json.h#L8

I have implemented this into our Theengs Decoder library for usage in several projects and also use it personally to deactivate/activate my alarm system when I'm coming home/leaving the house, only when the Apple Watch worn and unlocked by me.

And yes, the behaviour of how to prevent locking when on the charger is similar to my alarm system not being set when a firmware update presents the Watch as absent for quite a few minutes ;)  my suggestion would be a MUTE/PAUSE option in the BLEUnlock menu, with a very clear icon indicator to the pause.

ts1, 2024-04-19:
@DigiH Thank you for the very interesting information. 🙏 
I will look into your project more closely. 👀