ts1/BLEUnlock #123: [Feature Request] Multiple BLE Devices needed to unlock/lock

chunthebear, 2024-03-14:
Having multiple devices adds an additional layer of security. E.g., my phone and watch. 

(Really appreciate the work!)

squatto, 2024-08-21:
This is a duplicate of #11

timtrace, 2024-10-24:
> This is a duplicate of #11

#11 presumably asks for multiple devices to be allowed to unlock the Mac, with only one device being necessary. This topic seems to be about _requiring_ a connection with multiple devices to unlock.

squatto, 2025-01-03:
> #11 presumably asks for multiple devices to be allowed to unlock the Mac, with only one device being necessary. This topic seems to be about _requiring_ a connection with multiple devices to unlock.

Ah, I think you're right. Good catch!