ts1/BLEUnlock #125: Is installing BLEUnlock an open door to the security of the Mac?

jesusdeveloper, 2024-03-21:
Hello!

I've used it a lot because I love BLEUNLOCK, until today...

For example, I used the Apple Watch as a remote device. But...
What happens if I leave the watch at home and a family member approaches the Apple Watch (Apple Watch with a locked security code). Access the Mac without a password!!!!!!

This seems quite insecure to me, I say in the house, but let's imagine that you use it in an office and you forget the watch in the office someday...

**Is there anything for BLEUnclock to check that the Apple Watch is unlocked before unlocking the Mac?**

Thank you, best regards!

ts1, 2024-03-21:
Hello,
Checking Apple Watch unlocked seems possible according to #113.
I'll investigate it when I have time.
In addition, using BLEUnlock itself increases security risk.
See #90.
I'll describe it in README when I have time.

jesusdeveloper, 2024-03-21:
Wow, thank you very much, well, it's a good thing I've noticed. It's sad because bleunlock is very comfortable!!! :)
But safety comes first. What the user proposes in #113 seems great to me, I wish it could be done sometime.

Regards and thank you for your answer!!

sevitz, 2024-03-21:
I use apple native to unlock my laptop (as they requires the watch to be unlocked).

But I use BLEunlock to lock the device. This makes it more secure, without any compromise.