ts1/BLEUnlock #140: macOS 15 Sequoia compatibility

Jimbolino, 2024-09-27:
It doesnt appear to be able to unlock my screen automatically after updating to macOS 15
Perhaps something on the lock screen has changed, and the fakeKeyStrokes are not received, or some timing is off.

the-mace, 2024-10-27:
Im on Sequoia and it does unlock my screen as expected.

Jimbolino, 2024-10-27:
> Im on Sequoia and it does unlock my screen as expected.

Weird, which device are you using? I'm on a MacBooks air M1. Maybe it's getting too old/slow after the Sequoia update

the-mace, 2024-10-27:
Sorry should have put that in. This is a M2 Macbook Air so shouldnt be age of the machine with the settings below.
[image]
Using my iPhone 16 Pro to lock/unlock. 
The whole behavior of it (other than the apps quitting) is wonderful.

3x071c, 2024-10-29:
I'm on Sequoia and just set up BLEUnlock. Can confirm that Wake on Proximity isn't working either (I have unlock disabled). My external monitor wakes up (from idle/no input source) when approaching, but the screen stays black.

Zane-Zzx, 2024-11-18:
> I'm on Sequoia and just set up BLEUnlock. Can confirm that Wake on Proximity isn't working either (I have unlock disabled). My external monitor wakes up (from idle/no input source) when approaching, but the screen stays black.

Me too. I use my Android phone. It can lock after I leave for a while, but the unlock function didn't work. Even worse, when I unlock my Mac manually, it locks suddenly after a few minutes and I have to unlock it by myself again.